vicocz/brickcontroller2 #52: icons in branch "release/2024-09"

J0EK3R, 2024-11-01:
Hi @vicocz 😃 
I tried out your branch "release/2024-09" because of the QR-sharing.
When I run the app, none of the buttons had icons!?

I.e. the following code in CreationListPage.xaml
`<controls:ToolbarIcon Icon="qr_code_scanner" Text="{extensions:Translate ScanQr}" Command="{Binding ScanCreationCommand}" />`
Where is the icon "qr_code_scanner" located?

Btw.: QR-sharing works pretty well! ;)

vicocz, 2024-11-01:
Hi,

>When I run the app, none of the buttons had icons!?

Do you mean Windows application? It's known issue related to symbol fonts, probably located somewhere in Windows API. Saw some issues in MAUI related to that but they have been reported fixed, but nothing has changed. Android should be fine.

> Where is the icon "qr_code_scanner" located?
 I've applied symbol font from Google and `Icon` property refers to a glyph from that font. You can have a look e.g. [here](https://fonts.google.com/icons?icon.style=Rounded&icon.platform=web&icon.query=qr_code_scanner&selected=Material+Symbols+Rounded:qr_code_scanner:FILL@0;wght@400;GRAD@0;opsz@24&icon.size=24&icon.color=%235f6368)
The reason for this change was that I used to struggle finding icons for new features and this seems to me as easier way even though we are now limited with set of symbols available in that font.

>Btw.: QR-sharing works pretty well! ;)

Thx, I've tried to do it as smooth as possible :)

vicocz, 2024-11-01:
Anyway branch `release/2024-09` is still kind of temporary one, I still plan to review those commits before merging it to the default one. Just a warning :)
It's tough to maintin all the changes, especially if I tried to bricg most of them to the original repo, still neither merged nor declined.

J0EK3R, 2024-11-02:
> Anyway branch release/2024-09 is still kind of temporary one

I know - you've warned me...! ;)

> It's tough to maintin all the changes, especially if I tried to bricg most of them to the original repo, still neither merged nor declined.

Yeah, I'm watching the "progress" of the original...
It's... hmmm... the plant is growing slowly ;)

> Do you mean Windows application? It's known issue related to symbol fonts, probably located somewhere in Windows API. Saw some issues in MAUI related to that but they have been reported fixed, but nothing has changed. Android should be fine.

Your're right - Android is fine, Windows not.

J0EK3R, 2024-11-02:
Btw.: I've completely restructured [my repo](https://github.com/J0EK3R/brickcontroller2) with a lot of branches:

[master](https://github.com/J0EK3R/brickcontroller2/tree/master) -> original master branch from István Murvai's repo
[main](https://github.com/J0EK3R/brickcontroller2/tree/main) -> my default release branch
[/save/master](https://github.com/J0EK3R/brickcontroller2/tree/save/master) -> my old working branch
/feature/... -> a branch for each feature I'm implementing
[/feature/Merge_Vicocz](https://github.com/J0EK3R/brickcontroller2/tree/feature/Merge_Vicocz) -> your /release/2024-09 stuff merged into mine
[/merge/vicocz/release/2024-09](https://github.com/J0EK3R/brickcontroller2/tree/merge/vicocz/release/2024-09) -> branch for pull requests into your repo

J0EK3R, 2024-11-03:
I found the solution and created a [pull-request](https://github.com/vicocz/brickcontroller2/pull/55).

closed 2024-11-03 by J0EK3R