vicocz/brickcontroller2 #101: [Discussion] extend BC2 for other kinds of devices

J0EK3R, 2025-02-18:
Dear @vicocz :)

I'm very impressed about your work in this repo: your coding style is exemplary and the project grows very agile...

Freely tranlated: "I would like to bet on this horse" :)

I would really love it to extend your fork of BC2 to other kinds of devices!

In my mind there are at first position the BLE-Advertising devices I've implemented in my fork yet...

But there are others in my mind that could also profit from the controlablility by a gamepad.
I.E. the [microbit](https://microbit.org/) per [serial over Bluetooth](https://support.microbit.org/support/solutions/articles/19000062330-using-the-micro-bit-bluetooth-low-energy-uart-serial-over-bluetooth-).

Are you interested in principle?

Some rework would be necessary to handle a growing number of supported devices: the handling of the devices would have to be more modular.

I.E. the detection of devices in BluetoothDeviceManager.GetDeviceIfo should be in a specific factory class for each device or device group.

Or: why using the DeviceTypeToImageConverter - each device could return it's own image... would be more modular

What do you think about my idea?

J0EK3R, 2025-02-18:
I took a closer look at the code and - in my opinion - if the project should be modular the problem is the enumeration DeviceTypes and it's usages:

Every device has to be listed in that enumeration and each switch-case has to take care of each device.
When implementing new devices you have to change all that!

Imagine there is a common project BrickController2 and several seperate projects for each manufacturer "Lego", "MK", "Micro:Bit", etc. - you can't maintain that enumeration and all usages of it really good and it's not modular.

Why not just use a string for the device type?

---

Another part breaking the modularity is the DeviceTypeToImageConverter and it's colleague DeviceTypeToSmallImageConverter.

Why not extend the class Device with a Property Icon and send DeviceTypeToImageConverter and DeviceTypeToSmallImageConverter to retirement?

---

The tough nut to crack is DeviceChannelSelector.xaml.cs - I'm not an expert for that UI-stuff.
In this one xaml is a static UI part for each device.
In my opinion - the devices (or their factories/viewmodel?) should return their specific UI part.
Or the UI should be generic and controlled by device properties.

---

Without having the closest look on all usages of the enumeration DeviceType I would say replacing that would be the best approach to get modular...

vicocz, 2025-02-18:
Hi, general answer is YES. I guess it's more long term goal as adding support for BT device such as MK might take huge time to review such changes.
Related to `DeviceTypeToImageConverter` it's quite big topic as modularity here might be quite challenging. I thought in past BrickLink2 project should be split into non UI aka "Core" part and UI pieces. But it will create a lot of differences in case BC2 repository will be maintained again. So no clear anser here.

I think I need to see some branch to see examples of code you are talking about.

J0EK3R, 2025-02-20:
OK, let's freeze the reworking stuff and talk about getting the BLE-Advertising into the project.
...how things are currently done is working pretty well! ;)

---

BLE Advertising is a one-way communication - no scanning is possible - we need a dialog to select the devices to insert. I would suggest a second button next to the scan button to open the dialog.
To restore the devices on app-start the currently used mechanism using the enumeration DeviceType will work.

---

Another question is the naming - do we use the real name of MK and the others and what's about the images?
Currently I've taken same quick photos.

---

Next level is the c@d@ stuff - the telegrams to control the devices are BLE Advertising like MK.
But you need a kind of pairing.
But the pairing is a bit complicated: when scanning for new devices you have to advertise a special telegram containing a unique id. The devices are responding with another telegram containing their id.
The evaluation can be done in BluetoothDeviceManger like I've done [here](https://github.com/J0EK3R/brickcontroller2/blob/main/BrickController2/BrickController2/DeviceManagement/BluetoothDeviceManager.cs)

---

First I will start as simple as possible with the implementation of MK6.0 and perhabs a single button to insert the MK6.0 device.
I will create a single branch for that and will request a pull...

closed 2025-05-22 by J0EK3R