vicocz/brickcontroller2 #114: [Feature] add BLE Advertising capablility to iOS

J0EK3R, 2025-03-17:
Currently the BLE Advertising capability isn't implemented.

We need somebody with an iOS device, Mould King MK6.0 or MK4.0 and a bit of time...

@LarsDD0815 - interested?

LarsDD0815, 2025-03-17:
Basically, yes! Only, I have no Idea, what BLE Advertising is *lol*. What would be needed?

 But I do have an iOS device + MK 6 ;)

J0EK3R, 2025-03-17:
An MK device for testing, an iOS device, and the **ability to create the iOS app** are the basics.
And, of course, a bit of time ;)

Let's just try...

I need some time to prepare the basics.

...schreib mir doch mal, gerne auf Deutsch ;) an J0EK3R bei gmx Punkt net

J0EK3R, 2025-03-18:
I've created a [fork of this repo](https://github.com/J0EK3R/bc2) and invited @LarsDD0815 and @vicocz as collaborators to allow pushes.

...never done before, I hope it works

J0EK3R, 2025-04-01:
Haha! Here's what I discovered:

I revived my trusty iPad and installed the Mould King App on it to capture the Bluetooth telegrams sent...

And: `They look different than the ones I've seen on Android.`

`This means that the same Advertising Telegrams do not have to be generated as with the Android app.`

Android App:
[image]

iOS App:
[image]

vicocz, 2025-04-01:
> Haha! Here's what I discovered:
> 
> iOS App: [image]

It looks weired. This seems tobe totaly different adv packet, more the clasic one with advertised service uuids.
Are you sure you have captured datagram with some command (as obviously it's the one from Android) ?

J0EK3R, 2025-04-02:
Yes, I'm pretty shure to have captured the right datagram... ;)
I look at the raw data at the right - the fill-bytes
`12 13 14 15 16 17` are an indicator for MouldKing...
I think they `missuse` this kind of datagram because iOS and Windows doesn't allow to send the datagrams like in Android.
First I will try to send such datagrams from the Windows part of the App...

vicocz, 2025-04-02:
I see, I did not compare hex bytes as you did, nice Chinese hack 😄

J0EK3R, 2025-04-26:
I'm working on it - my iPad is too old (12.5.7) to enable developer mode so I ordered a newer one with FW >16.

Next week I'll try to build the project using VSCode or Rider and I hope to be able to deploy the App to the new iPad.

J0EK3R, 2025-05-14:
here is a small interim report

...I'm still working on it:
* I got the toolchain working: developing with VSCode on Windows, building on a virtual Mac, and deploying to a local iPad.
* I figured out how to create the advertising data from a raw byte array using the 16-bit service UUIDs

...and I'm afraid the encryption is a bit different from Android's.

vicocz, 2025-05-14:
What exactly do you mean encryption? Are you trying to reverse engenier communication od their IOS app?

Aby snippet od that raw building of adv data please? I'm just curious 😁

J0EK3R, 2025-05-14:
By *encryption* I mean encrypting the raw data in [MKProtocol.GetRfPayload](https://github.com/vicocz/brickcontroller2/blob/default/BrickController2/BrickController2/Protocols/MKProtocol.cs).

There is no docu available for the MK protocol - so I have to reverse engineer, yes.

I'm using Wireshark with a special Bluetooth dongle to capture the BT traffic and tools like Ghidra...

Code snippet is coming... ;)

J0EK3R, 2025-05-14:
Here is the snipped with some comments...

```C
using System;
using System.Linq;
using System.Threading.Tasks;
using CoreBluetooth;
using Foundation;
using BrickController2.PlatformServices.BluetoothLE;
using System.Collections.Generic;

namespace BrickController2.iOS.PlatformServices.BluetoothLE;

internal class BluetoothLEAdvertiserDevice : CBPeripheralManagerDelegate, IBluetoothLEAdvertiserDevice
{
    private CBPeripheralManager? _peripheralManager;
    private StartAdvertisingOptions? _advData;

    public BluetoothLEAdvertiserDevice()
    {
    }

    protected override void Dispose(bool disposing)
    {
        StopAdvertiseInternal();
        _peripheralManager?.Dispose();

        base.Dispose(disposing);
    }

    public Task StartAdvertiseAsync(AdvertisingInterval advertisingIterval, TxPowerLevel txPowerLevel, ushort manufacturerId, byte[] rawData)
    {
        SetAdvertisingData(manufacturerId, rawData);

        return Task.CompletedTask;
    }

    public Task StopAdvertiseAsync()
    {
        StopAdvertiseInternal();

        return Task.CompletedTask;
    }

    public Task UpdateAdvertisedDataAsync(ushort manufacturerId, byte[] rawData)
    {
        SetAdvertisingData(manufacturerId, rawData);

        return Task.CompletedTask;
    }

    private void SetAdvertisingData(ushort manufacturerId, byte[] manufacturerSpecificData)
    {
        byte[] header = {0xf9, 0x08, 0x49};    // bytes taken from Wireshark
        byte[] rawData = header.Concat(manufacturerSpecificData).ToArray();    // quick&dirty for the moment

        List<CBUUID> ser_list = new();
        for (int index = 0; index < rawData.Length; index += 2)
        {
            CBUUID _uuidService = CBUUID.FromBytes([    // a two-byte-array creates a 16-bit Service
                rawData[index + 1], 
                rawData[index + 0]]);

            ser_list.Add(_uuidService);
        }

        _advData = new StartAdvertisingOptions { ServicesUUID = ser_list.ToArray() };

        if (_peripheralManager?.Advertising == true)
        {
            _peripheralManager.StopAdvertising();
        }

        StartAdvertisingInternal();
    }

    private void StartAdvertisingInternal()
    {
        // Initialize peripheral manager if not already done.
        if (_peripheralManager == null)
        {
            _peripheralManager = new CBPeripheralManager();
            _peripheralManager.StateUpdated += (sender, e) =>
            {
                if (_peripheralManager.State == CBManagerState.PoweredOn && _advData != null)
                {
                    _peripheralManager.StartAdvertising(_advData);
                }
            };
        }

        if (_peripheralManager.State == CBManagerState.PoweredOn && _advData != null)
        {
            _peripheralManager.StartAdvertising(_advData);
        }
    }

    private void StopAdvertiseInternal()
    {
        _peripheralManager?.StopAdvertising();
        // Reset data so that it does not start advertising automatically if the interface goes ON.
        _advData = null;
    }
}
```

J0EK3R, 2025-05-15:
Yes, it's a hack from MK. And - as I've seen - the other manufacturers from China are doing things like MK.

I think the problem is that advertising like we're doing in the Android part of the BC2 app is not supported in iOS (and Windows). So this is a pretty hacky way of advertising any raw data one-way.

...just the result is counting... ;)

J0EK3R, 2025-05-20:
Here we are - I proudly present an working iOS advertiser for MK4 and MK6.

@LarsDD0815 - probier's gerne mal aus :)

https://github.com/vicocz/brickcontroller2/pull/133

PR #137 (2025-05-30): Enable BLE advertising on IOS
Seems this was omitted in #133.

Resolves #114.
closed 2025-05-30 by vicocz