vicocz/brickcontroller2 #119: [Feature] add ByteArrayExtension.CompareValues(byte[], byte[])

J0EK3R, 2025-03-22:
What do you think about creating an extension in `ByteArrayExtension` to compare byte arrays.

We could reduce code linke

```
 if (outputValues[0] != _lastOutputValues[0] || 
   outputValues[1] != _lastOutputValues[1] || 
   outputValues[2] != _lastOutputValues[2] || 
   outputValues[3] != _lastOutputValues[3] || 
   sendAttemptsLeft > 0)
```
to
```
 if (!outputValues.CompareValues(_lastOutputValues) || 
   sendAttemptsLeft > 0)
```
I know that we could use 
```
outputValues.SequenceEqual(_lastOutputValues)
```
but there is pretty much overhead inside (i.e. working with enumerators).

Or do you have another suggestion?

vicocz, 2025-03-22:
No strong opinion on that, `SequenceEqual` does not look as bad as you have described it. Any extension method should have some checks, such as nulls and comparsion of array requires e.g. comparsion of lengths. So not sure if thats the difference against [SequenceEqual](https://github.com/dotnet/runtime/blob/b3788239f221a29fa58ffeec54b3639458113a4c/src/libraries/System.Linq/src/System/Linq/SequenceEqual.cs#L11).

May be you can use this example to create such extension method:
```
public bool AsSpanSequenceEqual(int[] firstArray, int[] secondArray)
{
    return firstArray.AsSpan().SequenceEqual(secondArray);
}
```

vicocz, 2025-03-22:
I suggest to ad such extension methods and it's implementation can be finetuned later.

J0EK3R, 2025-03-22:
OK.
I think the extension don't need to be very common - it could be very special, i.e. no null-checks, no size-checks, only comparing byte arrays with the same size.
Else let's use SequenceEquals...

vicocz, 2025-04-23:
Resolved, there is `bool SequenceEqual<TItem>(this TItem[]? x, TItem[]? y)` extension method in `BrickController2.Extensions`, recently added. It uses span comparisin which should be effecient enough.

closed 2025-04-23 by vicocz