vicocz/brickcontroller2 #159: the PC version needs easier installations

EricKidson, 2025-08-02:
Recently, i was on PC and install appx x64 file in recent and i got lots of errors when install

i hope i will replace WITH .zip file for easier install

J0EK3R, 2025-08-02:
I tested it on my Win11 PC.

First thing is the warning about a certificate...

[image]

This is the full german text
> Das Herausgeberzertifikat dieses App-Pakets konnte nicht verifiziert werden.Wenden Sie sich an Ihren Systemadministrator oder den App-Entwickler, um ein neues App-Paket mit verifizierten Zertifikaten zu erhalten. Das Stammzertifikat und alle Sofortzertifikate der Signatur im App-Paket müssen verifiziert werden (0x800B010A)

Translated to English:
> The publisher certificate of this app package could not be verified. Contact your system administrator or the app developer to obtain a new app package with verified certificates. The root certificate and all instant signing certificates in the app package must be verified (0x800B010A).

J0EK3R, 2025-08-05:
@vicocz - would't it be better to have an installer without this certificate stuff?

vicocz, 2026-08-17:
@EricKidson Hi, yes, I missed its expiration, there should be the new one applied for the latest Beta release.