vicocz/brickcontroller2 #164: Add Model Context Protocol (MCP) Server

J0EK3R, 2025-08-16:
A few days ago, I read a very interesting article about AI and the Model Context Protocol (MCP).

This gave rise to the idea of extending BC2 with the capabilities of an MCP server.

The MCP server would (like the game controller does) simply control channels in BC2 with the values (-1..0..1).

That would be pretty cool – you could use the app in an educational setting to get closer to the topic of AI.

The setting could be:
* [Claude's desktop app](https://claude.ai/download)
* [MCP server that integrates a USB camera](https://github.com/groundlight/framegrab-mcp-server) that films overhead a playing field with a model car
* BC2 with the MCP server to control the model car

You could then give the AI tasks like:
* Drive a circle
* Drive a figure eight
* Drive to the points
* Follow the line

What do you think?

J0EK3R, 2025-08-22:
**HowTo**

I've pushed PR #165 `[POC] McpServer for Win`.

> 💡 **Tip:** If you want to try it out just get my [branch feature/MCPServer](https://github.com/J0EK3R/brickcontroller2/tree/feature/MCPServer) to your PC and build the Win App...

Now, if the `Brickcontroller2 Win App` is started, the `McpServer` is shown in the `Controller tester` view as `Controller`:

[image]

You have to add the (bridge script to connect to the) McpServer in the `claude_desktop_config.json` like this

```
{
  "mcpServers": {
    "brickcontroller": {
      "command": "py",
      "args": [
        "C:\\Projects\\brickcontroller2\\BrickController2\\MCPServer\\mcp_http_bridge.py",
        "--url",
        "http://localhost:5000"
      ]
    }
  }
}
```
The bridge script `mcp_http_bridge.py` is located in the brickcontroller repo.

_In a later Release the Claude App would perhaps directly support MCPServers over http without using a wrapper._

Then just start the [Claude Desktop App](https://claude.ai/download) and prompt something:

First prompt `list tools` to enshure the brickcontroller MCPServer is available:

[image]

Next prompt could be `set channel 1 to 0,5 and channel 2 to -0,99`

[image]

You'll see the result in the `Controller tester`:

[image]

Next prompt could be `stop all channels`

[image]

vicocz, 2025-08-26:
Hi,
still not sure about it whether it's worth implementing it.
I guess the app is intended to be the simplest one. THerefore there is no UI to control the models but just gamepad. From that perspective this adds unnecessary complexity and increase codebase.
On the other hand any AI integration might be interesting to play with. But it would required more inputs IMHO, channel motor telemetry, sensor info, etc. Which is the thing this app is definitely not reeady for. I have long term plans (or better say dreams) to rewrite the app using different framework, but it's too far and too big to implement.

Currently I'm quite busy with preparation for some  LEGO exhibition and my priorities related to the app is to have better release process, ideally using the officially stores. Therefore I think there is no chance to invest my time to this feature.

I will reconsider it once thew exhibition is over. ONe possibility is to have some experimental branch, but I guess it's nearly the same as to keep your own fork with this feature.

Regards
Vit

J0EK3R, 2025-08-27:
Hello Vit,
Thanks for your open words and for sharing your dreams and plans (for the app) with me! ;)
Yes, the lack of time (for the things you really want) – that's my problem too.

In my opinion, implementing the MCP server could improve the app and is very useful, for example, in education – the buzzword AI is ubiquitous in the future – I'm using it more and more for programming.

The impact on the code is very minimal – in simple terms, it's just another controller that triggers events, and the option to enable it in the settings page (and set the port and password).

> But it would required more inputs IMHO, channel motor telemetry, sensor info, etc. Which is the thing this app is definitely not reeady for.

In my opinion, it's sufficient to act as a game controller; the AI just using BC2 as bridge to the motors will do the rest.

Here's my proposal: I'll implement the MCP server for all our platforms and prepare the code for merging.
The impact on the existing codebase will really be minimal.
If you're ready, we'd be happy to discuss it again.

...that's one of my dreams... ;)